osresearch/spispy #15: Monitor mode

osresearch, 2019-06-27:
The monitor mode should be able to receive incoming data on MOSI to log both sides of the transactions.

emard, 2019-06-27:
I'm not into spi flashes much but for monitoring,
there's interesting project for multichannel 
storage oscilloscope that works on ULX3S

It's standalone (PC not needed to view or control), runs off BRAM only
(doesn't use SDRAM), displays colored traces as realtime video on the digital monitor
and has windowed GUI which is controlled by PS/2 mouse connected to
US2 port (over OTG cable and/or suitable USB/PS2 adapter)

main site is this:
https://github.com/hdl4fpga/hdl4fpga

I have a fork with experimental features
https://github.com/emard/hdl4fpga

osresearch, 2019-06-28:
Love the idea of using the video output as a scope.... Perhaps that will be useful in a later project!

The goal for this patch was to add some basic protocol debugging to the SPI bus, which worked well enough just by dumping the bytes to the FTDI port, not quite as ambitious.

osresearch, 2019-06-28:
Monitor mode works; needs to be selected at build time.  A better user interface #3 should make it runtime selectable.

closed 2019-06-28 by osresearch