osresearch/spispy #30: Understanding vulnerable BIOS landscape

jans23, 2020-02-27:
From reading about SpiSpy I understand that the vulnerability exploited by SpiSpy has been fixed by Intel and/or BIOS vendors. Of course the fix may not be distributed to all affected computers in the field. But I'm wondering if SpiSpy works with older systems only or are still vulnerable systems coming on the market?

osresearch, 2020-02-27:
To the best of my knowledge no vendors have deployed the mitigations; the staging branch has not been merged into the mainline edk2 and Intel continues to discuss it on their bugzilla page. A few days ago this update was posted: https://bugzilla.tianocore.org/show_bug.cgi?id=1614#c10

> After further evaluation on a few platforms, the solution provided in edk2-staging is not preferred for wider deployment because it requires a high number of platform firmware touchpoints. This is primarily because all pointers into CPU LLC can not automatically be migrated if the pointers were stored in custom data structures in dynamically allocated buffers. These could be updated by platform-specific firmware code with knowledge of the data buffer address and data structure format but the process is time consuming and all execution paths that might access a pointer to the CAR address range must be analyzed. The changes also increased complexity of core components such as PeiCore which is undesirable.

There are many, many additional TOCTOU in the various vendor firmwares that need to be addressed.  The plan to unmap the SPI flash makes good sense as a way to wipe out this class of vulnerability, but the chance of the vendors actually fixing them seems low.

jans23, 2020-02-27:
Thank you for clarification.

closed 2020-02-27 by jans23