osresearch/spispy #32: Missing LICENSE

TechnologyClassroom, 2020-11-10:
I see you have no LICENSE file for this project. The default is copyright.

I would suggest releasing the code under the GPL-3.0-or-later or AGPL-3.0-or-later license so that others are encouraged to contribute changes back to your project.

mithro, 2020-11-10:
Please don't use AGPL.

GPL or Apache 2.0 would fine.

TechnologyClassroom, 2020-11-10:
@mithro Why?

mithro, 2020-11-10:
It doesn't make much sense for HDL code and [means I can't touch it](https://opensource.google/docs/using/agpl-policy/).

TechnologyClassroom, 2020-11-10:
I try to license all of my code under the AGPL because it is impossible to tell how code will interact with the Internet in the future.  It probably does not make sense for the project.

TechnologyClassroom, 2020-11-11:
What is the relationship between this project and Google?  Does Google contribute or fund the project?

mithro, 2020-11-11:
As far as I know, none at the moment, but it would prevent us for ever potentially funding this project in the future.

We do contribute to the open source FPGA toolchain and I have previously talked to @osresearch about a potential LiteX based version of this tool too.

osresearch, 2020-11-11:
Thanks for pointing out that a license file is missing, @TechnologyClassroom .  I think the GPL is the best choice for this project.  There are two major components that are not my copyright - the SDRAM controller that is 2-clause BSD, and scanlime's UART that is MIT.  Both of those are GPL compatible, and I'm not too worried about tivoization of the code, nor someone running hosted versions, so the extra protections of the GPLv3 are not necessary.

I've been meaning to join the LiteX and nMigen party, @mithro , but I'm still writing Verilog by hand like a caveperson. One of these days I'll put down the stone tools.

mithro, 2020-11-11:
FYI -- The preferred license for things in the LiteX / nMigen ecosystem is "BSD-2-Clause License".